SwiftGust/ardupilot_gazebo #6: Compilation error on Gazebo 9

johnbdk, 2018-11-18:
Hello,

Trying to build the project on gazebo-9, 'make' gave me the following error:

In file included from /usr/include/boost/variant.hpp:22:0,
                 from /usr/include/sdformat-6.1/sdf/Param.hh:24,
                 from /usr/include/sdformat-6.1/sdf/Element.hh:25,
                 from /usr/include/sdformat-6.1/sdf/sdf.hh:4,
                 from /home/j/ardupilot_gazebo/src/ArduCopterPlugin.cc:27:
/usr/include/boost/variant/get.hpp: In instantiation of ‘typename boost::add_reference<T>::type boost::strict_get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’:
/usr/include/boost/variant/get.hpp:284:25:   required from ‘typename boost::add_reference<T>::type boost::get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’
/usr/include/sdformat-6.1/sdf/Param.hh:296:31:   required from ‘bool sdf::Param::Get(T&) const [with T = short unsigned int]’
/usr/include/sdformat-6.1/sdf/Element.hh:454:7:   required from ‘std::pair<T, bool> sdf::Element::Get(const string&, const T&) const [with T = short unsigned int; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/usr/include/sdformat-6.1/sdf/Element.hh:429:55:   required from ‘T sdf::Element::Get(const string&) const [with T = short unsigned int; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/j/ardupilot_gazebo/src/ArduCopterPlugin.cc:55:12:   required from ‘bool getSdfParam(sdf::ElementPtr, const string&, T&, const T&, const bool&) [with T = short unsigned int; sdf::ElementPtr = std::shared_ptr<sdf::Element>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/j/ardupilot_gazebo/src/ArduCopterPlugin.cc:590:45:   required from here
/usr/include/boost/variant/get.hpp:212:5: error: static assertion failed: boost::variant does not contain specified type U, call to boost::get<U>(boost::variant<T...>&) will always throw boost::bad_get exception
     BOOST_STATIC_ASSERT_MSG(
     ^

Is there anything to do with the project itself or it is my own fault (like missing new version of a library or something like that) ?

SwiftGust, 2018-11-19:
Hello,
Have you installed libgazebo9-dev for additonal feature?

johnbdk, 2018-11-20:
Thanks for the reply. Unfortunately, I switched back to Gazebo 7, however I am planning to test it as soon as possible (I'll keep you informed).

galdalali, 2018-11-25:
I get the same make error too

SwiftGust, 2018-11-26:
@galdalali Thank you for testing out. 
It will take some time to take a look at it since I don't have a clean desktop to test it,
meanwhile could you stay at gazebo 7 or 8?

SwiftGust, 2018-11-29:
@johnbdk @galdalali 
Others say installing gazebo-9 branch on Ubuntu 18.04 fine.
https://discuss.ardupilot.org/t/arducopter-3-6-sitl-with-gazebo-unstable-flight/34451/12 

It looks this is caused by old Boost Library, very common library used in everywhere,
and commonly in ROS, Gazebo too.
Ubuntu tends to have not up-to-date boost library,
please try to manually update Boost library or try it on Ubuntu 18.04
Then let me know if there's any more problem.

galdalali, 2018-11-29:
Yep, it works fine on Ubuntu 18.04.

SwiftGust, 2018-12-02:
Leave a comment if the problem still exists.

closed 2018-12-02 by SwiftGust