SwiftGust/ardupilot_gazebo #12: Installing the Environment with Gazebo 9

mustaphos, 2019-03-17:
Hi.
At the make -j4 command i get the following error:
```
Scanning dependencies of target ArduCopterIRLockPlugin
Scanning dependencies of target ArduCopterPlugin
Scanning dependencies of target ArduPilotPlugin
[ 16%] Building CXX object CMakeFiles/ArduCopterIRLockPlugin.dir/src/ArduCopterIRLockPlugin.cc.o
[ 33%] Building CXX object CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o
[ 50%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc: In member function ‘void gazebo::ArduPilotPlugin::OnUpdate()’:
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc:914:39: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’; did you mean ‘SetSimTime’?
     this->dataPtr->model->GetWorld()->GetSimTime();
                                       ^~~~~~~~~~
                                       SetSimTime
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc: In member function ‘void gazebo::ArduPilotPlugin::ApplyMotorForces(double)’:
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc:999:62: error: ‘class gazebo::physics::Joint’ has no member named ‘GetAngle’; did you mean ‘GetChild’?
         const double pos = this->dataPtr->controls[i].joint->GetAngle(0).Radian();
                                                              ^~~~~~~~
                                                              GetChild
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc: In member function ‘void gazebo::ArduPilotPlugin::SendState() const’:
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc:1195:53: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’; did you mean ‘SetSimTime’?
   pkt.timestamp = this->dataPtr->model->GetWorld()->GetSimTime().Double();
                                                     ^~~~~~~~~~
                                                     SetSimTime
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc:1245:27: error: ‘class gazebo::physics::Model’ has no member named ‘GetWorldPose’; did you mean ‘SetWorldPose’?
     this->dataPtr->model->GetWorldPose().Ign();
                           ^~~~~~~~~~~~
                           SetWorldPose
/home/m/ardupilot_gazebo/src/ArduPilotPlugin.cc:1278:38: error: ‘class gazebo::physics::Link’ has no member named ‘GetWorldLinearVel’; did you mean ‘WorldLinearVel’?
     this->dataPtr->model->GetLink()->GetWorldLinearVel().Ign();
                                      ^~~~~~~~~~~~~~~~~
                                      WorldLinearVel
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc: In member function ‘void gazebo::ArduCopterPlugin::OnUpdate()’:
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc:565:68: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’; did you mean ‘SetSimTime’?
   gazebo::common::Time curTime = this->dataPtr->model->GetWorld()->GetSimTime();
                                                                    ^~~~~~~~~~
                                                                    SetSimTime
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc: In member function ‘void gazebo::ArduCopterPlugin::SendState() const’:
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc:752:53: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’; did you mean ‘SetSimTime’?
   pkt.timestamp = this->dataPtr->model->GetWorld()->GetSimTime().Double();
                                                     ^~~~~~~~~~
                                                     SetSimTime
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc:802:27: error: ‘class gazebo::physics::Model’ has no member named ‘GetWorldPose’; did you mean ‘SetWorldPose’?
     this->dataPtr->model->GetWorldPose().Ign();
                           ^~~~~~~~~~~~
                           SetWorldPose
/home/m/ardupilot_gazebo/src/ArduCopterPlugin.cc:833:38: error: ‘class gazebo::physics::Link’ has no member named ‘GetWorldLinearVel’; did you mean ‘WorldLinearVel’?
     this->dataPtr->model->GetLink()->GetWorldLinearVel().Ign();
                                      ^~~~~~~~~~~~~~~~~
                                      WorldLinearVel
/home/m/ardupilot_gazebo/src/ArduCopterIRLockPlugin.cc: In member function ‘virtual void gazebo::ArduCopterIRLockPlugin::OnNewFrame(const unsigned char*, unsigned int, unsigned int, unsigned int, const string&)’:
/home/m/ardupilot_gazebo/src/ArduCopterIRLockPlugin.cc:237:14: error: ‘using element_type = class gazebo::rendering::Visual {aka class gazebo::rendering::Visual}’ has no member named ‘GetWorldPose’; did you mean ‘SetWorldPose’?
         vis->GetWorldPose().pos.Ign(), camera);
              ^~~~~~~~~~~~
              SetWorldPose
/home/m/ardupilot_gazebo/src/ArduCopterIRLockPlugin.cc:262:26: error: ‘using element_type = class gazebo::rendering::Visual {aka class gazebo::rendering::Visual}’ has no member named ‘GetName’; did you mean ‘SetName’?
       this->Publish(vis->GetName(), pt.X(), pt.Y());
                          ^~~~~~~
                          SetName
CMakeFiles/ArduPilotPlugin.dir/build.make:62: recipe for target 'CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o' failed
make[2]: *** [CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ArduPilotPlugin.dir/all' failed
make[1]: *** [CMakeFiles/ArduPilotPlugin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMakeFiles/ArduCopterPlugin.dir/build.make:62: recipe for target 'CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o' failed
make[2]: *** [CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/ArduCopterPlugin.dir/all' failed
make[1]: *** [CMakeFiles/ArduCopterPlugin.dir/all] Error 2
CMakeFiles/ArduCopterIRLockPlugin.dir/build.make:62: recipe for target 'CMakeFiles/ArduCopterIRLockPlugin.dir/src/ArduCopterIRLockPlugin.cc.o' failed
make[2]: *** [CMakeFiles/ArduCopterIRLockPlugin.dir/src/ArduCopterIRLockPlugin.cc.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/ArduCopterIRLockPlugin.dir/all' failed
make[1]: *** [CMakeFiles/ArduCopterIRLockPlugin.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
How can i solve this?
Thanks...

SwiftGust, 2019-03-18:
Did you tried install with Gazebo-9 branch?

Accessor functions starting with Get have been changed since Gazebo version 9

Can you change and retry with changing code a little bit with

```
cd ~/ardupilot_gazebo
git checkout gazebo9
```
and try to build & install

mustaphos, 2019-03-18:
Hi again.
I succesfully build with your suggestion.
But when i do:
`gazebo --verbose zephyr_ardupilot_demo.world`
Then i get this error:
```
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
[Err] [Node.cc:105] No namespace found
```
Thanks...

SwiftGust, 2019-03-18:
It is quite sure about location environment variable is wrong
`Error Code: 11 Msg: Unable to find uri[model://zephyr_delta_wing_ardupilot_demo]`

It is little bit different file structure regarding branch gazebo9,

I found the instruction is wrong

Variables should be 
```
export GAZEBO_MODEL_PATH=~/ardupilot_gazebo/gazebo_models:${GAZEBO_MODEL_PATH}
export GAZEBO_RESOURCE_PATH=~/ardupilot_gazebo/gazebo_worlds:${GAZEBO_RESOURCE_PATH}
```

Regarding gazebo9 branch,  it is not well updated since I don't use Gazebo9, sorry for the trouble.

mustaphos, 2019-03-18:
@SwiftGust now it works. I started apm planner and connected to the plane. Added a simple mission but all the modes are incorrectly mapped. And when i start the mission the zephyr plane immediately turns upside down.

SwiftGust, 2019-03-19:
@mustaphos as long as I tested it on gazebo 7 & 8, zephyr flew fine with default gains

It would be nice if you post a log

ArduPilot creates SITL log directory **/logs** under where you SITL launched

One thing to know is that the dynamics engine is subject to change, depending on Gazebo version

I think you might need to tune PID gains of your plane

mustaphos, 2019-03-19:
> @mustaphos as long as I tested it on gazebo 7 & 8, zephyr flew fine with default gains
>
> It would be nice if you post a log
>
> ArduPilot creates SITL log directory **/logs** under where you SITL [...]
[00000001.zip](https://github.com/SwiftGust/ardupilot_gazebo/files/2982367/00000001.zip)
You can find the logs in the zip file.

SwiftGust, 2019-03-19:
I don't get this log, this is full of nonsense

1. Desired roll went wrong from the beginning and none of the rotations are wrapped to -180 ~ 180 degree range, this could be a issue of log parsing of APMPlanner but not sure since I don't use ArduPlane often.
2. How auto - arm could be initiated even EKF is not using GPS.
3. With full actuation, attitude error does not goes to zero
is this coincide with gazebo visual of control surfaces?

[image]

P.S Making this offset little bit negative may help prop rotating from the start
https://github.com/SwiftGust/ardupilot_gazebo/blob/gazebo9/gazebo_models/zephyr_delta_wing_ardupilot_demo/delta_wing.sdf#L128

mustaphos, 2019-03-19:
I found the solution.
In autopilot side, Servo 1 function must be 78 (not 77) and Servo 2 function must be 77 (not 78).
It fixes the issue

SwiftGust, 2019-03-19:
@mustaphos Okay, that sounds great.

I'm closing this since problems are solved

If there is more issues please re-open this or open another one.

closed 2019-03-19 by SwiftGust