SwiftGust/ardupilot_gazebo #13: Multiple UAVs in One Simulation Environment

mustafagokce09, 2019-03-20:
Hi.
How can i simulate multiple copter of plane in one simulation environment.
In SITL part i can do this:
```
sim_vehicle.py -v ArduPlane -f gazebo-zephyr -I0
sim_vehicle.py -v ArduPlane -f gazebo-zephyr -I1
```
What should i do in the world file?
Thanks...

SwiftGust, 2019-03-21:
Look closely here https://github.com/SwiftGust/ardupilot_gazebo#multi-vehicle-simulation

You need world with multiple zephyr with different port connection, 
overwrite connection settings in the world or models 
https://github.com/SwiftGust/ardupilot_gazebo/blob/master/models/zephyr_delta_wing_ardupilot_demo/delta_wing.sdf#L134-L135

mustafagokce09, 2019-03-21:
I do not know whats happening but i created two models as `zephyr_delta_wing_ardupilot_1 ` and `zephyr_delta_wing_ardupilot_2` and also i changed their SYSIDs as 1 and 2. But for some time the second uav becomes unstable and it crashes. 
As you can see on the image i am loading same waypoints to them (takeoff and RTL) but second uav as you can see after the takeoff becomes unstable.
[image]

closed 2019-03-21 by mustafagokce09