SwiftGust/ardupilot_gazebo #14: Ardurover Ardupilot interface issue

rishab26201, 2019-05-22:
Hi,

I'm trying to interface Ardupilot and Husky rover in Gazebo. While running the commands given I always get the following error
`Failed to load plugin libhusky_gazebo_plugins.so: libhusky_gazebo_plugins.so: cannot open shared object file: No such file or directory`
and also 
`Conversion of sensor type[imu] not suppported`
I tried ignoring the above errors and continue running the program. However when moving from one point to another the rover doesn't move in a straight line but rather very strangely. I have attached a link to a video of the motion.

[https://drive.google.com/open?id=1F4ZfQmBAp3rbz9X7IEO-gn158GszYUQl](url)

Thanks

SwiftGust, 2019-05-27:
Hi, those errors irrelevant part for ArduRover, 
it is simply looking for original ROS plugin.

For the weird movement, I can't see your video link
but it looks like parameter problem.

please check ArduRover parameters are correctly set to skid-steering rover
If it does, you may need tune rover controllers.

closed 2019-06-14 by rishab26201

khanasif786, 2022-03-31:
I know many will come here after me. I was also facing problem in turns of husky model. The solution is to reduce the mass of base_link in husky_model, and it will solve the strange turning problem.

aaryanmurgunde, 2022-06-02:
> I know many will come here after me. I was also facing problem in turns of husky model. The solution is to reduce the mass of base_link in husky_model, and it will solve the strange turning problem.

You are a life saver man this fix worked like a charm :)
Thank you