SwiftGust/ardupilot_gazebo #15: Problem with setting up Gazebo 9 with SITL and Ardurover

sashankmodali, 2019-06-07:
Hi,
   While starting Sitl for APMrover2 with a world file in gazebo 9 , The rover starts moving in a specific direction even without setting up any script file. I'm only using the file, sim_vehicle.py  to start the simulation. 

 Could you tell me what is causing this issue? When I set the mass of the rover close to zero, it moves along one of the horizontal axes.

SwiftGust, 2019-06-10:
Hello,
I need more to be able to help
sim_vehicle.py with which argument are you using?
You should note that gazebo usually works with physics engine.

I think it is probably something wrong with friction but I need more specifics.
Is always moves along random axes, not to the specific direction? 
Is any of wheel turning? 
Mass of which have you modified? base link of husky? (which I don't recommend to modify it)
Have you increased mass then?

sashankmodali, 2019-06-11:
Hi, 
    I used the command : 
`sim_vehicle.py -v APMrover2 -f gazebo-rover -m --mav10 --console -I1`

The rover always moves in a specific direction depending on the mass. It's moving with the same speed but it changes direction quickly to that specific direction and then moves in a straight line. The wheels are turning. For the mass, yeah I modified that of the base link of husky.

sashankmodali, 2019-06-11:
Hi, please see this video for better understanding : 
https://drive.google.com/open?id=1-C1BXmSqwDAaMw9gm9Lpv9MvmPO_hLPW

SwiftGust, 2019-06-11:
@sashankmodali Okay, your video made me understood the problem right away.
It looks like these offsets are changed as ArduRover changes.

This is ArduPilot Gazebo interface setting problem in the model file.
Please modify offsets to 0.0. 
https://github.com/SwiftGust/ardupilot_gazebo/blob/master/models/rover_ardupilot/model.sdf#L109-L161
And check default parameter is set for skid-to-steer rover.

sashankmodali, 2019-06-12:
Hi. The offsets were actually 0.0 and I changed them back to -0.5 and they started working. It doesn't move on its own now. 

Thank you for your response.

closed 2019-06-13 by SwiftGust