SwiftGust/ardupilot_gazebo #19: Failed on retrieving camera data from gazebo to ros

elliotwave, 2020-01-01:
First of all it's more of a question on development instead of an issue. I am sorry if misplaced my question here as I think this is the closest I can get to source for asking this.

I was trying to retrieve the image data on iris quad copter so I can process it in ROS and return the command through mavros for simulation in gazebo. The topic seems to be publishing in gazebo topic but not in rostopic. I would think that the image topic would magically appear in rostopic so I can used it directly with ROS but it doesn't. On the ardupilot doc it is stated that I need ros_gazebo_camera package however I couldn't find it anywhere. 

I try to use [gazebo_ros_camera](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/jade-devel/gazebo_plugins/src/gazebo_ros_camera.cpp) plugin by compiling it and add it to .world file but no luck. I haven't try to dig in much deeper as I am an absolute beginner in gazebo plugin.
[image]
The closes I can get is to manually grab the gazebo topic data and try to convert it to cv matrix but it's not really effective and laggy.
```C
#include <gazebo/transport/transport.hh>
#include <gazebo/msgs/msgs.hh>
#include <gazebo/gazebo_client.hh>

#include <opencv2/imgproc/imgproc.hpp>
#include <opencv2/highgui/highgui.hpp>

#include <iostream>

/////////////////////////////////////////////////
// Function is called everytime a message is received.

    int width;
    int height;
    char *data;
    bool execute_once = false;
    cv::Mat A;

void cb(ConstImageStampedPtr &msg)
{
  if (execute_once == false){
    width = (int) msg->image().width();
    height = (int) msg->image().height();
    //+1 for null terminate
    execute_once = true;
  }

    data = new char[msg->image().data().length() + 1];
    memcpy(data, msg->image().data().c_str(), msg->image().data().length());
    cv::Mat image(height, width, CV_8UC3, data);
    cv::resize(image, A, cv::Size(320, 240), 0, 0, cv::INTER_AREA);

    cv::imshow("camera", A);
    cv::waitKey(1);
    delete data;  // DO NOT FORGET TO DELETE THIS, 
                  // ELSE GAZEBO WILL TAKE ALL YOUR MEMORY
}

/////////////////////////////////////////////////
int main(int _argc, char **_argv)
{
  // Load gazebo
  gazebo::client::setup(_argc, _argv);

  // Create our node for communication
  gazebo::transport::NodePtr node(new gazebo::transport::Node());
  node->Init();

  // Listen to Gazebo world_stats topic
  gazebo::transport::SubscriberPtr sub = node->Subscribe("~/iris_demo/iris_demo/gimbal_small_2d/tilt_link/camera/image", cb);

  // Busy wait loop...replace with your own code as needed.
  while (true)
    gazebo::common::Time::MSleep(10);

  // Make sure to shut everything down.
  gazebo::client::shutdown();
}
```
I also found this [repository](https://github.com/AurelienRoy/ardupilot_sitl_gazebo_plugin/blob/master/recorder_camera_plugin/src/recorder_camera_plugin.cpp) that has plugin to publish the camera data designed for ardupilot iris but it seems to be outdated as the last push has been 4 years ago and it seems that some files are missing when I try to compile it. Is there any work around so I can get the image data on rostopic? Any help would be much appreciated. Thank you

SwiftGust, 2020-01-02:
This repository does not support Gazebo-ROS integration in the current form, only establish connection between Gazebo-ArduPilot SITL.

If you want camera topic in ROS from Gazebo, you have to add gazebo-ros camera plugin connected to gazebo camera in the iris model file, 
For further details, refer gazebo wiki http://gazebosim.org/tutorials?tut=ros_gzplugins&cat=connect_ros

Also, other gazebo topics can be published in ros but will require appropriate plugin, you may need to write down your own plugin for some topics.

For Gazebo-ROS connected SITL, useful reference is PX4 SITL Gazebo repository https://github.com/px4/sitl_gazebo

elliotwave, 2020-01-15:
Thank you so much for clearing things out. It turns out I include the plugin in wrong section of different file. Suppose to put it in camera gimbal UDF file. It works now.

closed 2020-01-15 by elliotwave