SwiftGust/ardupilot_gazebo #20: How to use ardupilot_gazebo with ROS

anish-pratheepkumar, 2020-01-15:
Hai,

first of all Awesome work!

I have a doubt, here we are opening ardupilot_gazebo plugin without ROS. Can we actually open ardupilot_gazebo inside ROS so that I can run my autonomous control ros node and see the simulation ?

Thanks in advance!

SwiftGust, 2020-01-16:
If you are after just for control simulation you don't need Gazebo-ROS, 
you can use Gazebo<->ArduPilot as is then configure MAVROS node connected to ArduPilot SITL for ROS topics.

However If you need sensor topic in ROS such as Camera, Lidar etc(No IMU or GPS, this can be acquired from MAVROS) then it is different story, you should use Gazebo-ROS package.
You have to add gazebo-ros plugins in the current model.

Refer official Gazebo Guide regarding this, 
http://gazebosim.org/tutorials?tut=ros_installing&cat=connect_ros
http://gazebosim.org/tutorials?cat=connect_ros

anish-pratheepkumar, 2020-01-16:
Good Day, 
Thank you for your reply!
I have already installed Gazebo-ROS. 
Yes, I would like to access lidar, camera and so on by running MAVROS nodes. Since the goal is to do autonomous navigation in a non-GPS setup, using Opticalflow(PX4flow). 
So do you mean if I:-
1. initialize MAVROS
2. initialize GAZEBO-ROS
3. initialize GAZEBO-Ardupilot SITL separately (not with ros wrapping, then how to get this Ardupilot SITL connected with Gazebo-ROS?)
4. also how to connect MAVROS to GAZEBO-ROS-Ardupilot after doing 3rd step.
I am a bit confused here

Also I was trying to set up GAZEBO-Ardupilot. Unfortunately, I am getting an error in 'make -j4' step. I tried both khancyr and yours(SwiftGust).  I have Gazebo 9 is that an issue? As per this statement from your wiki "_ArduPilot setup for SITL launch Gazebo version 7.x or 8.x   The gazebo9 branch will works on gazebo >= 9.x"_
does it mean I have to just replace the numerical 7 or 8 with 9? 
Note: I will post the error I am getting while doing "make -j4" soon, when I get back to my working location. 
Thanks & regards!

anish-pratheepkumar, 2020-01-16:
Please find the error while running "make -j4". Do you have info as to why this error is occurring ?

anish@studienarbeit:~/ardupilot_gazebo/build$ make -j4
[ 16%] Building CXX object CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o
[ 33%] Building CXX object CMakeFiles/ArduPilotPlugin.dir/src/ArduPilotPlugin.cc.o
[ 50%] Building CXX object CMakeFiles/ArduCopterIRLockPlugin.dir/src/ArduCopterIRLockPlugin.cc.o
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc: In member function ‘void gazebo::ArduCopterPlugin::OnUpdate()’:
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:565:68: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’
   gazebo::common::Time curTime = this->dataPtr->model->GetWorld()->GetSimTime();
                                                                    ^
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc: In member function ‘void gazebo::ArduCopterPlugin::SendState() const’:
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:752:53: error: ‘class gazebo::physics::World’ has no member named ‘GetSimTime’
   pkt.timestamp = this->dataPtr->model->GetWorld()->GetSimTime().Double();
                                                     ^
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:802:27: error: ‘class gazebo::physics::Model’ has no member named ‘GetWorldPose’
     this->dataPtr->model->GetWorldPose().Ign();
                           ^
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:833:38: error: ‘class gazebo::physics::Link’ has no member named ‘GetWorldLinearVel’
     this->dataPtr->model->GetLink()->GetWorldLinearVel().Ign();
                                      ^
In file included from /usr/include/boost/variant.hpp:22:0,
                 from /usr/include/sdformat-6.2/sdf/Param.hh:26,
                 from /usr/include/sdformat-6.2/sdf/Element.hh:25,
                 from /usr/include/sdformat-6.2/sdf/sdf.hh:4,
                 from /home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:27:
/usr/include/boost/variant/get.hpp: In instantiation of ‘typename boost::add_reference<T>::type boost::strict_get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’:
/usr/include/boost/variant/get.hpp:284:25:   required from ‘typename boost::add_reference<T>::type boost::get(boost::variant<T0, TN ...>&) [with U = short unsigned int; T0 = bool; TN = {char, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, long unsigned int, unsigned int, double, float, sdf::Time, ignition::math::v4::Color, ignition::math::v4::Vector2<int>, ignition::math::v4::Vector2<double>, ignition::math::v4::Vector3<double>, ignition::math::v4::Quaternion<double>, ignition::math::v4::Pose3<double>}; typename boost::add_reference<T>::type = short unsigned int&]’
/usr/include/sdformat-6.2/sdf/Param.hh:300:31:   required from ‘bool sdf::Param::Get(T&) const [with T = short unsigned int]’
/usr/include/sdformat-6.2/sdf/Element.hh:454:7:   required from ‘std::pair<T, bool> sdf::Element::Get(const string&, const T&) const [with T = short unsigned int; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/usr/include/sdformat-6.2/sdf/Element.hh:429:55:   required from ‘T sdf::Element::Get(const string&) const [with T = short unsigned int; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:55:12:   required from ‘bool getSdfParam(sdf::ElementPtr, const string&, T&, const T&, const bool&) [with T = short unsigned int; sdf::ElementPtr = std::shared_ptr<sdf::Element>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/anish/ardupilot_gazebo/src/ArduCopterPlugin.cc:590:45:   required from here
/usr/include/boost/variant/get.hpp:212:5: error: static assertion failed: boost::variant does not contain specified type U, call to boost::get<U>(boost::variant<T...>&) will always throw boost::bad_get exception
     BOOST_STATIC_ASSERT_MSG(
     ^
CMakeFiles/ArduCopterPlugin.dir/build.make:62: recipe for target 'CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o' failed
make[2]: *** [CMakeFiles/ArduCopterPlugin.dir/src/ArduCopterPlugin.cc.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/ArduCopterPlugin.dir/all' failed

SwiftGust, 2020-01-16:
Are you using Gazebo 9?
Some of API changed in Gazebo 9 and working plugin for that is at gazebo-9 branch
I need to update master branch but didn't have time to do it...