SwiftGust/ardupilot_gazebo #21: Ardupilot_gazebo with ArduPlane not working

thorstenbar, 2020-01-18:
Hello,

i am try to get this plugin running with gazebo9 and ArduPlane4.0. But when i follow the instructions it is giving me always the same mistake: 
`[Err] [ArduPilotPlugin.cc:682] [zephyr_delta_wing_demo] imu_sensor [zephyr_delta_wing_demo::zephyr_delta_wing::zephyr/imu_link::imu_sensor] not found, abort ArduPilot plugin.` What is the problem here? Am i the only one? With ArduCopter and the iris model it is working fine, but i need ArduPlane.

SwiftGust, 2020-01-20:
Hi,
The zephyr flyingwing model should work fine and the model in this repo has IMU attached.
It seems you have another zephyr delta wing model that gazebo uri handler finds which does not have IMU like this one https://bitbucket.org/osrf/gazebo_models/src/default/zephyr_delta_wing/model.sdf.
 
Check your GAZEBO_MODEL_PATH or gazebo client for duplication.

thorstenbar, 2020-02-16:
Hey,
i found the other version at /usr/share/gazebo-9/models and removed it. Thank you!

closed 2020-04-28 by SwiftGust

jlunsan, 2022-06-17:
I am having the same issue with ArduPlane and Gazebo and not finding the solution. I have already checked  all the commented in https://github.com/SwiftGust/ardupilot_gazebo/issues/21#issuecomment-576088885a

I am following all te instructions and getting always the same problems:

[Wrn] [ArduPilotPlugin.cc:741] [zephyr_delta_wing_demo] imu_sensor scoped name [zephyr_delta_wing_demo::zephyr_delta_wing::zephyr/imu_link::imu_sensor] not found, trying unscoped name.

[Err] [ArduPilotPlugin.cc:753] [zephyr_delta_wing_demo] imu_sensor [zephyr_delta_wing_demo::zephyr_delta_wing::zephyr/imu_link::imu_sensor] not found, abort ArduPilot plugin.

AlgorhtymicFatalError, 2022-12-28:
> I am having the same issue with ArduPlane and Gazebo and not finding the solution. I have already checked all the commented in [#21 (comment)](https://github.com/SwiftGust/ardupilot_gazebo/issues/21#issuecomment-576088885a)
> 
> I am following all te instructions and getting always the same problems:
> 
> [Wrn] [ArduPilotPlugin.cc:741] [zephyr_delta_wing_demo] imu_sensor scoped name [zephyr_delta_wing_demo::zephyr_delta_wing::zephyr/imu_link::imu_sensor] not found, trying unscoped name.
> 
> [Err] [ArduPilotPlugin.cc:753] [zephyr_delta_wing_demo] imu_sensor [zephyr_delta_wing_demo::zephyr_delta_wing::zephyr/imu_link::imu_sensor] not found, abort ArduPilot plugin.

hi did you solve it. im getting the same error. Thank you