SwiftGust/ardupilot_gazebo #22: Boat simulation in ardupilot SITL with gazebo

imgod26, 2020-04-22:
Since model or world for boat is not available in this directory. I have the required model and world but don't know  where to copy it ardupilot_gazebo and then how to run it in a way such that SITL movements are reflected in gazebo. I am quite new to this and would really appreciate any help possible. 

First Command :
sim_vehicle.py -v APMrover2 -f sailboat  -m --mav10 --map --console

SwiftGust, 2020-04-23:
Currently my repository doesn't have surface or underwater vehicles, and I'm not familiar with surface vehicle dynamics.

If you wanna build boat model for ArduPilot you can use hydrodynamics plugin.
This is supported by gazebo simulator natively, docs at http://gazebosim.org/tutorials?tut=hydrodynamics&cat=physics.

And there is gazebo boat model for PX4 at https://github.com/PX4/sitl_gazebo for quick look, it doesn't use hydrodynamics plugin, it is simulating boat dynamics with own built plugin as in https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_usv_dynamics_plugin.cpp

imgod26, 2020-04-25:
Is there a way to use the worlds and models from this. https://bitbucket.org/osrf/vrx/src/default/ 

And run it with SwiftGust and Ardurover sailboat SITL.

SwiftGust, 2020-04-27:
It seems no problem at all using gazebo models & worlds in the VRX.

The only missing thing from VRX is USV model built with USV plugin in their own repository,
you can refer PX4 boat model, now I see it is same USV plugin from VRX, so this model dynamics should be compatible with VRX model - https://github.com/PX4/sitl_gazebo/blob/master/models/boat/boat.sdf

Other thing you require is to replace PX4 - Gazebo plugin setup with ArduPilot - Gazebo plugin in this repository.
PX4 Gazebo plugin settings seems quite large but nothing much since PX4 prefers modular architecture so you wouldn't have much problem.
https://github.com/PX4/sitl_gazebo/blob/0823f5e006011d36858c2dd39881874c1edac10b/models/boat/boat.sdf#L448-L577 except just leave usv dyanmics part at https://github.com/PX4/sitl_gazebo/blob/0823f5e006011d36858c2dd39881874c1edac10b/models/boat/boat.sdf#L515-L536

Regarding setting up ArduPilot - Gazebo Plugin, refer models in my repository 
example - https://github.com/SwiftGust/ardupilot_gazebo/blob/7da690d5612b6b6064a7ceb8d21c1add179663b1/models/zephyr_delta_wing_ardupilot_demo/delta_wing.sdf#L132-L186

Sensors - https://github.com/SwiftGust/ardupilot_gazebo/blob/7da690d5612b6b6064a7ceb8d21c1add179663b1/models/zephyr_delta_wing/model.sdf#L101-L188

Ric-27, 2020-07-13:
did you managed to use vrx models with ardupilot-gazebo plugin?
I was able to combine the ardupilot plugin with the vrx simulation (waves and wind) but when I click the ardupilot plugin inside gazebo i get the following error.
[image]

ericjohnson97, 2020-10-06:
I was able to setup a simulation using the ardupilot plugin and the bouyancy plugin from px4. here is my simulation repo. there is a section about setting it up for simulating boats https://github.com/Intelligent-Quads/iq_sim