SwiftGust/ardupilot_gazebo #23: Error Using ArduPilotPlugin With URDF Models

ed-muthiah, 2020-06-06:
I am currently trying to convert the Iris model.sdf to an urdf so that I can get RViz visualizations when I add the realsense camera plugin. I was able to successfully convert the Iris.sdf to and urdf to get the correct collisions, visuals and inertias.
[image]

However I'm having trouble with the ardupilot plugin. I am able to establish a connection but sending takeoff commands does not work. The command is acknowledge but nothing happens in Gazebo.
[image]

This the current urdf:
[model.txt](https://github.com/SwiftGust/ardupilot_gazebo/files/4739371/model.txt)

SwiftGust, 2020-06-08:
Hi,
I'm sorry but I don't get what you are trying to do.
How did you launched ArduPilot SITL with which command? 
Are you sure you are running  Gazebo-ArduPilot SITL?
Can you confirm vehicle attitude changes as it should be when you apply moment in gazebo side?

URDF is only useful in ROS side, using URDF to visualize robot in RVIZ is another thing to setup in RViz. You don't need collisions, inertial parameters to visualize. 
I see you made a URDF version of iris that gazebo can convert(using gzsdf) but I don't think it is neccessary to create Gazebo compatible URDF file.

ed-muthiah, 2020-06-08:
Hi @SwiftGust. FYI the normal SDF iris model and gazebo work fine for me. But I am trying to make some changes as follows.

*My goal*: Add realsense depth camera URDF+ Gazebo Plugin to Iris in Gazebo **AND visualise robot model + depth cloud in RViz**. 
*My issue*: RViz does not visualise SDF only URDF. To get point clouds in RViz you need the state to be published from an URDF robot for example, when I added the realsense gazebo plugin to an URDF rover I was able to get a point cloud like so:
[image]

With the iris drone in your repo, because it is defined in SDF, I cannot get a visualisation of the Robot Model or Depth Cloud in RViz. Like the above image. So I tried to convert the Iris SDF to an URDF then add the ardupilot plugin (and then depth camera plugin). However I cannot get this to run successfully. The URDF defined drone in gazebo connects to ardupilot but it does not take off.

In simple terms, _is it possible to define the Iris + Ardupilot Gazebo Plugin in URDF_?  

Here is someone else with a very similar problem: https://github.com/khancyr/ardupilot_gazebo/issues/26

I have sent you a message on gitter if you want to chat in more detail.

SwiftGust, 2020-06-09:
Hi @ed-muthiah.
Thank you for detailed explanation.
It seems you are having no issue attaching depth camera to iris model and getting depth image & pointcloud from the depth camera as ROS topic.

To visualize your robot in RViz, you don't need to setup anything more in Gazebo side.
You just need to load robot model and setup tf frames in ROS side alone as described here
http://wiki.ros.org/rviz/DisplayTypes/RobotModel

On the other hand, if you want to use URDF in Gazebo.
In the end you need SDF file for Gazebo simulation.
so the **gzsdf** command will give you check functionality URDF & conversion to SDF. If Gazebo doesn't work properly as described in http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros

So I think you should check this converted SDF file is correct in regarding plugin definition as same as in this repository.

hamsterasu, 2021-08-12:
Hi @ed-muthiah , I took a look at your model.txt file and found that there's an error at all the rotor joints, i.e
`<limit effort="0.0" lower="-1e+16" upper="1e+16" velocity="0.0"/>`. The maximum effort and velocity is 0.0, means they can't spin at all. I changed this line at all rotor joints to:
`<limit effort="2000.0" lower="-1e+16" upper="1e+16" velocity="838.0"/>`, and the drone was able to take off. 

However, there are still some weird behaviour like when initialising the gyro, it takes a little long, whereas in the sdf version it was almost instantaneous. 
[image]

Sometimes, the gyro does not even get initialised and I get this error :
`gyro[1] did not converge: diff=0.192727 dps ardupilot` . When this happens, the drone can take off but it will keep flying up forever.

It is no doubt a progress on what you've done though. Now, I just hope someone else has the answer has the answer to the gyro problem