SwiftGust/ardupilot_gazebo #25: How to simulate a custom tilt rotor with Gazebo

jzattelman, 2020-07-21:
I've created a custom tilt rotor model in gazebo and can successfully link it with the gazebo SITL. The model is a quadplane (which will have a wing in the future) and each motor arm is able to tilt forwards to  transition to fixed wing flight.

The bit that I'm stuck on is how to link the motor arm rotation joints to a channel so that I can control the angular position of each motor.

This is what I have currently:
**In my gazebo_tiltrotor.parm file:**
SERVO5_FUNCTION, 41

**In my model.sdf file....**
<control channel="5">
        <type>POSITION</type>
        <offset>0</offset>
        <p_gain>0.20</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.5</cmd_max>
        <cmd_min>-2.5</cmd_min>
        <jointName>iris::base_link_to_motor_arm_1_joint</jointName>
      </control>

How do I get Gazebo to take the PWM command from Ardupilot SITL and turn it into motor arm angular position? (eg 0 degrees is when the motor arm is vertical and 90 degrees is when the motor arm is tilted forward for fixed wing flight)

Thanks in advance!

SwiftGust, 2020-07-21:
Hi, 
Are you referring quad-tilt-rotor type aircraft? 
I think zephyr ardupilot model would be good example in this case since there is propeller & elevon.
I can't find good example with such aircraft in Gazebo

Refer this part of the section, https://github.com/SwiftGust/ardupilot_gazebo/blob/master/models/zephyr_delta_wing_ardupilot_demo/delta_wing.sdf#L132-L186

Use position type for tilt joint, and set other parameters correct such as offset, min & max by trial and error.
For the rough example, if upright position is 0 degree, then control output from ArduPilot without tilt command(i.e. PWM low) should make tilt joint to hold upright too. This ArduPilot control to gazebo joint command interpretation is up to multiplier, offset, cmd_min & cmd_max plugin parameters.

sundexin111, 2021-04-30:
is there a ArduPlanePlugin,my arduplane is not connected with gazebo

SwiftGust, 2021-04-30:
@sundexin111 ArduPilotPlugin is generic plugin for ArduPilot-Gazebo SITL, use this plugin and check your settings if it doesn't connect.
Please open another issue to discuss issue.