SwiftGust/ardupilot_gazebo #26: Plugin for Quadplane in Ardupilot?

aabs7, 2020-08-25:
Is there a plugin for quadplane for ardupilot? Can we use ardupilot plugin to control quadplane?

SwiftGust, 2020-08-26:
This is general plugin that communicates with ArduPilot, you can build your own quadplane gazebo model.

aabs7, 2020-08-26:
I've taken model of standard vtol from [px4/sitl_gazebo/models](https://github.com/PX4/sitl_gazebo/tree/master/models) and i changed the file for ardupilot plugin. I cannot get the vtol moving. 

To run SITL i've used parameters of quadplane that comes default in Arduplane i.e 
```
"quadplane": {
                "waf_target": "bin/arduplane",
                "default_params_filename": "default_params/quadplane.parm",
            }
```
- Is it the correct parameters to load into gazebo model? Because I can see that the model which can successfully connect to gazebo has a prefix 'gazebo-' in it like "gazebo-iris", "gazebo-zephyr". 

- Is there something to keep in mind while connecting ardupilot sitl to gazebo model using ardupilot plugin?

SwiftGust, 2020-08-28:
I don't get what you mean by "get the vtol moving", 
Issue could come from various reasons, do you get proper sensor reading, got EKF initialized? can you make the actuator moving with motor test?

parameter is depends on the airframe, loading default SITL parameter would not do harm.

shovan777, 2020-08-31:
@aabs7 and I are working in the same project. I did everything you told @SwiftGust in[ here ](https://github.com/prokuranepal/ardupilot_gazebo/blob/vtol/models/standard_vtol/standard_vtol.sdf). We were able to connect to the ardupilot sitl with the libardupilotplugin and we made the standard vtol able to work in quad mode. But we could not get the puller to work. I have assigned puller to [channel 2](https://github.com/prokuranepal/ardupilot_gazebo/blob/4aec44075a365865e517853e6f0bdaf354a4227a/models/standard_vtol/standard_vtol.sdf#L1184), as I thought since the channel start at 0 in the plugin and throttle is at 3 in plane, so it must be at 2. But it did not work and transition did not occur from quad to plane mode. Did I make a mistake in channel assignment? Please help!!!

SwiftGust, 2020-09-01:
@aabs7 do you have any screen recording of the issue? I don't get what you mean by puller, if it is tilt servo then you should set it position type actuator in your model sdf.
And also input channel and output channel is different in ArduPilot, you should set correct servo#_function* for proper channel mapping, you can utilize motor test from QGC or Mission Planner for testing.

shovan777, 2020-09-01:
Yes, finally got it. The main issue was the servo#_function which was not set properly. Should I make a pull request? @SwiftGust

SwiftGust, 2020-09-01:
@shovan777 It's good to hear you finally made it. PR is not mandatory for this repository but If you do, I would appreciate it.

shovan777, 2020-09-01:
Just made a pull request. Please verify that. @SwiftGust But the simulation also requires to create a gazebo-quadplane.parm file where servoN_functions are written. It would be great if gazebo-quadplane.parm had been in ardupilot repo.

shovan777, 2020-09-02:
Hi @SwiftGust, Can you tell us how we can add gazebo-quadplane.parm or gazebo-vtol.parm files to the ardupilot repo? These changes are as you know, essential for the simulation to work with ardupilot sitl.

SwiftGust, 2020-09-02:
Hi @shovan777 you can directly make PR on https://github.com/ardupilot/ardupilot for the Gazebo SITL parameters, script.

shovan777, 2020-09-04:
Hi @aabs7, I just made all the changes you requested in the pull #27 , please review them, and inform me if any other changes are reuquired.

SwiftGust, 2020-09-11:
As PR #27 is merged thanks to @shovan777, now this repository contains Quadplane.
Please test it and close the issue if your issue is resolved @aabs7.

aabs7, 2020-09-13:
The issue is resolved. Closing it now. Thanks @shovan777 @SwiftGust

closed 2020-09-13 by aabs7