SwiftGust/ardupilot_gazebo #28: Tiltrotor simulation error

shovan777, 2020-10-13:
We tried to do vtol tiltrotor simulation.  But we could not get it to work. We posted a detail of the problem [here](https://discuss.ardupilot.org/t/simulation-of-tiltrotor-in-gazebo/62260)

shovan777, 2020-10-13:
Hi @SwiftGust  can u help us with this? Are we making a mistake in parameters?

SwiftGust, 2020-10-13:
Hi,
Usual servo channel PWM min/max range is 1000, 2000 microseconds each respectively and actual range is narrower.
If your problem is propellers are spinning before arming the plane, then set parameter ARMING_REQUIRE 2 and see if it works.
https://ardupilot.org/plane/docs/parameters.html#arming-require-require-arming-motors

aabs7, 2020-10-14:
Hi @SwiftGust 
I've tried using ARMING_REQUIRE 2 but the problem persists. However, If I change the channels of motors to , 5, 6, 7, & 8 then the spinning stops. Is there some reason behind it ?
- Another problem with tiltrotor in gazebo that I've found is that, when the tilting motor is set as fixed joint when designing from solidworks then the model flies just like normal quad and there is no problem. However, when the tilting motor is set as revolute joint, and is controlled from ardupilot then, the joint doesn't move as per PWM value given by ardupilot. For example: the PWM value given to the revolute joint shows to be 1900 each (for motor to be vertical while taking off), But the tilting motor oscillates (Very ODD). What might be the reason of it?
- Another question is - when we make model for ardupilot from solidworks, and after we connect the model to ardupilot and see the mavlink message for vibration, We can see lots of vibration in x,y,z frame. But If we look at the vibration of Iris or zephyr, there is no vibration. How can we minimize vibration from design?

SwiftGust, 2020-10-16:
Hi @aabs7, to answer your questions

- Servo channels(for aileron, elevator, and so on) are free to move even it is disarmed, you may had set it to these channels instead of the rotor, so you should check servo*_channels' function you set to.

- If you find tilt servo oscilates you should set correct PID gains for actuators in your gazebo model, most of actuators are subject to control(think about the ESC, not all ESCs are same) and also gazebo models too. https://github.com/prokuranepal/ardupilot_gazebo/blob/tiltrotor/models/tilt_rotor/tilt_rotor.sdf#L882-L887

- Most of vibration in Gazebo model comes from model to ground plane collision(contact), check collision parameter for baselink or landing gears and use high damping ratio for IMU joint as seen iris or zephyr model.