SwiftGust/ardupilot_gazebo #29: Ardupilot and Gazebo11?

AJRepo, 2020-11-02:
Is there a way to get ardupilot to integrate with Gazebo11?

SwiftGust, 2020-11-03:
Hi @AJRepo 

AFAIK, there is no gazebo API change since Gazebo 9 so this repo would work with Gazebo 11 with 
`sudo apt install libgazebo11-dev`
Can you test this repository with Gazebo 11?
Let me know if you find a problem.

AJRepo, 2020-11-03:
Thanks!

I tried with Gazebo-11 installed
```
$ dpkg --list | grep -i gazebo
ii  gazebo11                                           11.1.0-1~focal                              amd64        Open Source Robotics Simulator
ii  gazebo11-common                                    11.1.0-1~focal                              all          Open Source Robotics Simulator - Shared files
ii  gazebo11-plugin-base                               11.1.0-1~focal                              amd64        Open Source Robotics Simulator - base plug-ins
ii  libgazebo11:amd64                                  11.1.0-1~focal                              amd64        Open Source Robotics Simulator - shared library
ii  libgazebo11-dev:amd64                              11.1.0-1~focal                              amd64        Open Source Robotics Simulator - Development Files
ii  ros-noetic-gazebo-dev                              2.9.1-1focal.20200724.170657                amd64        Provides a cmake config for the default version of Gazebo for the ROS distribution.
ii  ros-noetic-gazebo-msgs                             2.9.1-1focal.20200929.231255                amd64        Message and service data structures for interacting with Gazebo from ROS.
ii  ros-noetic-gazebo-plugins                          2.9.1-1focal.20200930.000612                amd64        Robot-independent Gazebo plugins for sensors, motors and dynamic reconfigurable components.
ii  ros-noetic-gazebo-ros                              2.9.1-1focal.20200929.235447                amd64        Provides ROS plugins that offer message and service publishers for interfacing with Gazebo through ROS.
ii  ros-noetic-gazebo-ros-control                      2.9.1-1focal.20200930.000638                amd64        gazebo_ros_control
ii  ros-noetic-gazebo-ros-pkgs                         2.9.1-1focal.20200930.003513                amd64        Interface for using ROS with the Gazebo simulator.```
 and make failed with both the the dev and master branches.  

Here's the output from `git checkout dev && mkdir build && cd build && cmake .. && make -j4`

[output_of_dev_branch_make.txt](https://github.com/SwiftGust/ardupilot_gazebo/files/5482189/make_dev.txt)

SwiftGust, 2020-11-03:
Hi, Gazebo 9 support is on master & gazebo-9 branches only, please try it on master branch and upload the error if you find trouble.

AJRepo, 2020-11-03:
Thanks!

Make failed with the master branch too. Here's the output after `git checkout master && make clean && cmake .. && make -j4`

[output_of_master_make.txt](https://github.com/SwiftGust/ardupilot_gazebo/files/5482660/output_of_master_make.txt)

lz82db, 2021-01-03:
I'm also trying to use the plug-in with Gazebo 11. The make command failed with the same message as AJRepo posted. If you have any guidance, I'm all ears.

rottaran, 2021-02-17:
same here on macos with clang:
```
In file included from /Users/rrotta/projekte/upwards/ardupilot_gazebo/src/ArduPilotPlugin.cc:17:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/functional:504:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:662:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/tuple:1067:5: error: 
      static_assert failed due to requirement 'value != __not_found' "type not
      found in type list"
    static_assert(value != __not_found, "type not found in type list" );
    ^             ~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/tuple:1080:14: note: 
      in instantiation of template class
      'std::__1::__find_detail::__find_exactly_one_checked<unsigned short, bool,
      char, std::__1::basic_string<char>, int, unsigned long long, unsigned int,
      double, float, sdf::v9::Time, ignition::math::v6::Angle,
      ignition::math::v6::Color, ignition::math::v6::Vector2<int>,
      ignition::math::v6::Vector2<double>, ignition::math::v6::Vector3<double>,
      ignition::math::v6::Quaternion<double>, ignition::math::v6::Pose3<double>
      >' requested here
    : public __find_detail::__find_exactly_one_checked<_T1, _Args...> {
             ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/variant:1492:24: note: 
      in instantiation of template class
      'std::__1::__find_exactly_one_t<unsigned short, bool, char,
      std::__1::basic_string<char>, int, unsigned long long, unsigned int,
      double, float, sdf::v9::Time, ignition::math::v6::Angle,
      ignition::math::v6::Color, ignition::math::v6::Vector2<int>,
      ignition::math::v6::Vector2<double>, ignition::math::v6::Vector3<double>,
      ignition::math::v6::Quaternion<double>, ignition::math::v6::Pose3<double>
      >' requested here
  return _VSTD::get_if<__find_exactly_one_t<_Tp, _Types...>::value>(__v);
                       ^
/usr/local/Cellar/sdformat9/9.5.0/include/sdformat-9.5/sdf/Param.hh:318:25: note: 
      in instantiation of function template specialization
      'std::__1::get_if<unsigned short, bool, char,
      std::__1::basic_string<char>, int, unsigned long long, unsigned int,
      double, float, sdf::v9::Time, ignition::math::v6::Angle,
      ignition::math::v6::Color, ignition::math::v6::Vector2<int>,
      ignition::math::v6::Vector2<double>, ignition::math::v6::Vector3<double>,
      ignition::math::v6::Quaternion<double>, ignition::math::v6::Pose3<double>
      >' requested here
        T *value = std::get_if<T>(&this->dataPtr->value);
                        ^
/usr/local/Cellar/sdformat9/9.5.0/include/sdformat-9.5/sdf/Element.hh:511:29: note: 
      in instantiation of function template specialization
      'sdf::v9::Param::Get<unsigned short>' requested here
      this->dataPtr->value->Get<T>(result.first);
                            ^
/usr/local/Cellar/sdformat9/9.5.0/include/sdformat-9.5/sdf/Element.hh:486:36: note: 
      in instantiation of function template specialization
      'sdf::v9::Element::Get<unsigned short>' requested here
    std::pair<T, bool> ret = this->Get<T>(_key, result);
                                   ^
/Users/rrotta/projekte/upwards/ardupilot_gazebo/src/ArduPilotPlugin.cc:69:39: note: 
      in instantiation of function template specialization
      'sdf::v9::Element::Get<unsigned short>' requested here
    _param = _sdf->GetElement(_name)->Get<T>();
                                      ^
/Users/rrotta/projekte/upwards/ardupilot_gazebo/src/ArduPilotPlugin.cc:954:3: note: 
      in instantiation of function template specialization 'getSdfParam<unsigned
      short>' requested here
  getSdfParam<uint16_t>(_sdf, "fdm_port_in",
```

This fails with sdformat9 version 9.5.0, installed via homebrew.

According to their [https://github.com/osrf/sdformat/blob/f59a342469bd9c71bd2dc5b325d777164375b06a/include/sdf/Param.hh#L308](Param.hh) , the supported types are only:
```
bool, char, std::string, int, std::uint64_t,
unsigned int, double, float, sdf::Time,
ignition::math::Angle,
ignition::math::Color,
ignition::math::Vector2i,
ignition::math::Vector2d,
ignition::math::Vector3d,
ignition::math::Quaterniond,
ignition::math::Pose3d
```

rottaran, 2021-02-17:
manually fixed this as follows:
```
diff --git a/src/ArduPilotPlugin.cc b/src/ArduPilotPlugin.cc
index f058996..1283ab5 100644
--- a/src/ArduPilotPlugin.cc
+++ b/src/ArduPilotPlugin.cc
@@ -951,10 +951,13 @@ bool ArduPilotPlugin::InitArduPilotSockets(sdf::ElementPtr _sdf) const
       this->dataPtr->fdm_addr, "127.0.0.1");
   getSdfParam<std::string>(_sdf, "listen_addr",
       this->dataPtr->listen_addr, "127.0.0.1");
-  getSdfParam<uint16_t>(_sdf, "fdm_port_in",
-      this->dataPtr->fdm_port_in, 9002);
-  getSdfParam<uint16_t>(_sdf, "fdm_port_out",
-      this->dataPtr->fdm_port_out, 9003);
+  unsigned tmp;
+  getSdfParam<unsigned>(_sdf, "fdm_port_in",
+      tmp, 9002);
+  this->dataPtr->fdm_port_in = tmp;
+  getSdfParam<unsigned>(_sdf, "fdm_port_out",
+      tmp, 9003);
+  this->dataPtr->fdm_port_out = tmp;
 
   if (!this->dataPtr->socket_in.Bind(this->dataPtr->listen_addr.c_str(),
       this->dataPtr->fdm_port_in))
```

but probably the correct way would be to change the type of fdm_port_in etc?

BV-OpenSource, 2021-03-11:
Made a Pull Request #34 that resolved this problem for me.

SwiftGust, 2021-03-26:
@rottaran @BV-OpenSource Thank you for debugging & resolve issue.
PR for resolve this issue is merged now.

Please re-open this issue If anyone encounter same issue.

closed 2021-03-26 by SwiftGust