SwiftGust/ardupilot_gazebo #30: APM: EKF2 IMU0 ground mag anomaly, yaw re-aligned when run with gazebo_ros

varunvp, 2020-11-27:
Hello,
Until 2 weeks ago the setup that I was using with Arducopter 3.6.7, Gazebo 9 and ROS was working fine. But from then the drone, after giving a takeoff command, takes off with a tilt, gives the following errors,

```
ARMED
Got COMMAND_ACK: NAV_TAKEOFF: ACCEPTED
APM: EKF2 IMU0 ground mag anomaly, yaw re-aligned
APM: EKF2 IMU1 ground mag anomaly, yaw re-aligned
APM: GPS Glitch
APM: EKF variance
LAND> Mode LAND
```

and flips in air and crashes.

[image]

This seems to be a ROS specific problem because the same world when run without ROS behaves normally. This problem happens when I run with gazebo_ros node. I don't understand how this could be interfering with Gazebo's IMU. I'm getting the same problem with khancyr's plugin.

varunvp, 2020-11-27:
I have uploaded the autopilots status after start up in each of the cases. I can see some differences in AHRS values. They are higher in the case of ROS. Same in the case of roll and yaw.
[StatusWithoutROS.txt](https://github.com/SwiftGust/ardupilot_gazebo/files/5606425/StatusWithoutROS.txt)
[StatusWithROS.txt](https://github.com/SwiftGust/ardupilot_gazebo/files/5606426/StatusWithROS.txt)

SwiftGust, 2020-12-03:
Hi,
It is unclear it is related to this repository. I don't personally tested this repository with ROS.
Thus problem cannot be reproduced on my side
You need to describe the details how to reproduce your issue if you need help.
Can you confirm you are using proper ArduPilot config for MAVROS & MAVLink?

varunvp, 2020-12-03:
I'll send the instructions. I'm using the default configs, I haven't changed anything. Can you tell me which configs you are referring to?

SwiftGust, 2020-12-03:
@varunvp Please post instruction here for further reference for others.

I wonder which launch & config file did you use? https://github.com/mavlink/mavros/tree/master/mavros/launch
MAVLink message interpretation & usage ArduPilot & PX4 can be quite different in detail.

closed 2021-01-17 by varunvp