SwiftGust/ardupilot_gazebo #31: QuadPlane Rotors Spinning up during Initialisation

Spartian-Omega, 2020-12-16:
I have made a simple Quadplane model and have configured the channels in the ArduPilot correctly for my model.

I have noticed when the ArduPilot SITL is initialising all my rotors spin up. This wouldn't be an issue, however once initialisation has finished the rotors are still spinning. I was able to rectify the motors spining after initialisation by using the offset Param, but this offset actually adds onto that initial motor spin up and causes the QuadPlane to takeoff.

Does anyone know anything about this?

Is it a Parameter setting in ArduPlane or is it a plugin thing?

SwiftGust, 2020-12-16:
@Spartian-Omega 
Hi, it sounds like output channel is mixed up, servo*_function parameters are not correctly set-up according to model definition - https://github.com/SwiftGust/ardupilot_gazebo/blob/master/models/standard_vtol/standard_vtol.sdf#L1151-L1260
Check parameters here - https://github.com/SwiftGust/ardupilot_gazebo/pull/27#issuecomment-687682056

Imagine it is just same as putting servo output channels from flight controller(pixhawk or else) to motor channels.

Spartian-Omega, 2020-12-16:
I will have to cross check the files you provided, but quickly, does say, servo#1 match with channel#1, or does it map to channel#0.

Spartian-Omega, 2020-12-17:
@SwiftGust I just checked the files you linked and according to them I have configured the channels and servos correctly in my case. So maybe there's something else causing the rotors to spin up during ArduPilots initialisation?

SwiftGust, 2020-12-17:
@Spartian-Omega I think most likely the issue would be related to this [parameter](https://ardupilot.org/plane/docs/parameters-Plane-stable-V4.0.4.html?highlight=disarm#arming-require-require-arming-motors).
Please check this parameter, PWM output for motor channels would not be 0 for ArduPlane.

If problem persists check followings

- Have you reverted change you made to the gazebo model? 
- Have you changed channel's trim?

Spartian-Omega, 2020-12-18:
@SwiftGust thanks for your reply. I checked the parameter you suggested and it is set to 1, this may have been causing my issue. However, I verified and ensured that I was using your version of the plugin and now that I am I am no longer experiencing the issue (rotors are no longer spinning up during initialization). 

The issue may have been caused by using khancyr version of the plugin?

I wasn't aware that there were differences between the two repositories plugins but after checking some other posts I am inclined to think that there is.

If that is the case, I am happy for you to close this issue.

SwiftGust, 2021-01-28:
HI @Spartian-Omega 
Sorry for keep you waiting, I was away for about a month for some reason.

As I review the code of khancyr's version, I see nothing much of difference,
most of difference is how to get parameters from the SDF.
However, khancyr's version does not support Gazebo version 9 since Gazebo API is changed,
which would not be able to compile with Gazebo 9 alone(without lesser version installed)

Which version of gazebo are you using?

Spartian-Omega, 2021-01-28:
HI again @SwiftGust 

I am using Gazebo 9 and I don't believe I have a previous version hidden away and it compiled fine a ran without throwing any errors. His documentation is confusing as it has instructions for running with ROS-Melodic with Gazebo 9.

Regardless changing to your version has eliminated that error, suggesting his documentation is contradicting itself.

SwiftGust, 2021-02-19:
Hi @Spartian-Omega.
I think I should address remove old version in the Readme later.
Then Is this issue resolved?

Spartian-Omega, 2021-03-03:
Hi, encountered the issue again, found the cause this time. The cause isn't your plugin it is in SIM_Gazebo.cpp function Gazebo::send_servos(...). If a servo is being commanded 0, it is being subtracted by 1000 leading to the value being passed to the ArduPilot Gazebo Plugin being -1000/1000 = -1. For QuadPlane the default functionality when disarmed is to disable the output to the QAssist motors, effectively sending them 0. 

A work around in the mean time is to change a QuadPlane parameter: [Q_SAFE_DISARM](https://ardupilot.org/plane/docs/parameters.html#q-m-safe-disarm-motor-pwm-output-disabled-when-disarmed) = 0 (from 1) to enable motor output while disarmed and to ensure [Q_M_PWM_MIN](https://ardupilot.org/plane/docs/parameters.html#q-m-pwm-min-pwm-output-miniumum) is still 1000.

An additional Plane parameter which may help as @SwiftGust pointed out is to ensure [ARMING_REQUIRE](https://ardupilot.org/plane/docs/parameters.html#arming-require-require-arming-motors) = 1, so that the minimum throttle value is set as the motors output while the craft is disarmed. 

Additionally check [SERVO_(X)_MIN](https://ardupilot.org/plane/docs/parameters.html#servo2-min-minimum-pwm) for the motor throttles, I have found that Q_M_PWM_MIN overwrites these for the QAssist motors.

This issue has been resolved here thanks for your help. I am unsure if this is worth requesting a change on the ArduPilot repo as I am unsure how adversely it will effect the functionality of other models.

closed 2021-03-03 by Spartian-Omega