SwiftGust/ardupilot_gazebo #37: Deltawing crashes at takeoff

apolitof, 2021-08-03:
Hi,
I am trying to make the Delta Wing fly with SITL and Gazebo. I installed Ardupilot 4.0.9 and Gazebo 9. I installed the ardupilot gazebo plugin at the branch "gazebo 9". I can start SITL and Gazebo without problems. After everything starts, I give the mission "CMAC-circuit.txt", then I set in AUTO mode and I arm the throttle. The Delta Wing starts, it reaches a couple of meters altitude, then turns at 180° roll and falls on the ground. No error is printed on screen. I had a look at all the parameters but I found nothing that could cause the problem. Did I miss something or did I do something wrong?

apolitof, 2021-08-03:
These are the parameters from mav.parm:

`ACRO_LOCKING     0.000000
ACRO_PITCH_RATE  180.000000
ACRO_ROLL_RATE   180.000000
ADSB_ENABLE      0.000000
AFS_ENABLE       0.000000
AHRS_COMP_BETA   0.100000
AHRS_CUSTOM_PIT  0.000000
AHRS_CUSTOM_ROLL 0.000000
AHRS_CUSTOM_YAW  0.000000
AHRS_EKF_TYPE    2.000000
AHRS_GPS_GAIN    1.000000
AHRS_GPS_MINSATS 6.000000
AHRS_GPS_USE     1.000000
AHRS_ORIENTATION 0.000000
AHRS_RP_P        0.200000
AHRS_TRIM_X      0.000000
AHRS_TRIM_Y      0.000000
AHRS_TRIM_Z      0.000000
AHRS_WIND_MAX    0.000000
AHRS_YAW_P       0.200000
ALT_CTRL_ALG     0.000000
ALT_HOLD_FBWCM   0.000000
ALT_HOLD_RTL     10000.000000
ALT_OFFSET       0.000000
ARMING_ACCTHRESH 0.750000
ARMING_CHECK     1.000000
ARMING_MIS_ITEMS 0.000000
ARMING_REQUIRE   1.000000
ARMING_RUDDER    1.000000
ARSPD_FBW_MAX    22.000000
ARSPD_FBW_MIN    13.000000
ARSPD_TYPE       0.000000
AUTOTUNE_LEVEL   6.000000
AUTO_FBW_STEER   0.000000
AVD_ENABLE       0.000000
BATT2_MONITOR    0.000000
BATT3_MONITOR    0.000000
BATT4_MONITOR    0.000000
BATT5_MONITOR    0.000000
BATT6_MONITOR    0.000000
BATT7_MONITOR    0.000000
BATT8_MONITOR    0.000000
BATT9_MONITOR    0.000000
BATT_AMP_OFFSET  0.000000
BATT_AMP_PERVLT  17.000000
BATT_ARM_MAH     0.000000
BATT_ARM_VOLT    0.000000
BATT_BUS         0.000000
BATT_CAPACITY    3300.000000
BATT_CRT_MAH     0.000000
BATT_CRT_VOLT    0.000000
BATT_CURR_PIN    12.000000
BATT_FS_CRT_ACT  0.000000
BATT_FS_LOW_ACT  0.000000
BATT_FS_VOLTSRC  0.000000
BATT_LOW_MAH     0.000000
BATT_LOW_TIMER   10.000000
BATT_LOW_VOLT    0.000000
BATT_MONITOR     4.000000
BATT_SERIAL_NUM  -1.000000
BATT_VOLT_MULT   10.100000
BATT_VOLT_PIN    13.000000
BATT_WATT_MAX    0.000000
BRD_ALT_CONFIG   0.000000
BRD_BOOT_DELAY   0.000000
BRD_OPTIONS      0.000000
BRD_PWM_COUNT    8.000000
BRD_RTC_TYPES    1.000000
[... 1120 lines cut ...]
TKOFF_THR_MINACC 0.000000
TKOFF_THR_MINSPD 0.000000
TKOFF_THR_SLEW   0.000000
TKOFF_TIMEOUT    0.000000
TRIM_ARSPD_CM    1600.000000
TRIM_AUTO        0.000000
TRIM_PITCH_CD    0.000000
TRIM_THROTTLE    45.000000
TUNE_CHAN        0.000000
TUNE_CHAN_MAX    2000.000000
TUNE_CHAN_MIN    1000.000000
TUNE_ERR_THRESH  0.150000
TUNE_MODE_REVERT 1.000000
TUNE_PARAM       0.000000
TUNE_RANGE       2.000000
TUNE_SELECTOR    0.000000
USE_REV_THRUST   2.000000
WP_LOITER_RAD    60.000000
WP_MAX_RADIUS    0.000000
WP_RADIUS        90.000000
YAW2SRV_DAMP     0.000000
YAW2SRV_IMAX     1500.000000
YAW2SRV_INT      0.000000
YAW2SRV_RLL      1.000000
YAW2SRV_SLIP     0.000000`

SwiftGust, 2021-09-23:
Hi,
You don't have to use gazebo-9 branch anymore, should use master branch for any gazebo version.

I suspect control failure due to ill-tuned control gain.
Posting a ArduPilot log can be helpful.

SwiftGust, 2021-10-11:
Hi, @apolitof 
for SITL you can find SD card like contents at where you started **sim_vehicle.py** script, check **logs directory**

SwiftGust, 2021-10-17:
Hi @apolitof, @ShivamSyal2k @NDevDrone 

It looks like plane went into stall condition, try to get further airspeed before get into maneuver.
You should know that more the bank angle in vehicle requires more airspeed to maintain lift required to keep vehicle in the air.
As far as I recall, zephyr model is little bit under-powered to maintain steep maneuver, try to keep airspeed more than 15m/s in level flight.

So my suggestion is following

1. Try to increase thrust by increase propeller speed like 838 to 1000 or more in https://github.com/SwiftGust/ardupilot_gazebo/blob/master/models/zephyr_delta_wing_ardupilot_demo/delta_wing.sdf#L148
2. Try to set parameter https://ardupilot.org/plane/docs/stall-prevention.html and tune airspeed controller

SwiftGust, 2021-11-04:
HI @apolitof  @NDevDrone  @ShivamSyal2k 
Is this issue still persists after applying suggested solutions?

NDevDrone, 2021-12-01:
@SwiftGust No, it took a bit of investigation to figure out what the issue was. Thankfully we found a set up. Took two changes.

1) After set up in full, validate gazebo delta wing loads but doesn't fly right. Then navigate to the SITL vehicle params folder at ~/ardupilot/Tools/autotest/default_params, and edit the file that appears here named 'gazebo-zephyr.parm' to have the following contents.

SERVO1_FUNCTION = 77
SERVO2_FUNCTION = 78
SERVO3_FUNCTION = 70
SERVO1_REVERSED = 0
SERVO2_REVERSED = 0
SERVO3_REVERSED = 0
SERVO1_MAX = 1900
SERVO1_MIN = 1100
SERVO1_TRIM = 1500
SERVO2_MAX = 1900
SERVO2_MIN = 1100
SERVO2_TRIM = 1500
SERVO2_MAX = 1900
SERVO2_MIN = 1100
SERVO2_TRIM = 1100
# we need small INS_ACC offsets so INS is recognised as being calibrated
INS_ACCOFFS_X   0.001
INS_ACCOFFS_Y   0.001
INS_ACCOFFS_Z   0.001
INS_ACCSCAL_X   1.001
INS_ACCSCAL_Y   1.001
INS_ACCSCAL_Z   1.001
INS_ACC2OFFS_X   0.001
INS_ACC2OFFS_Y   0.001
INS_ACC2OFFS_Z   0.001
INS_ACC2SCAL_X   1.001
INS_ACC2SCAL_Y   1.001
INS_ACC2SCAL_Z   1.001
INS_GYR_CAL      0
ARSPD_TYPE 0
BATT_MONITOR    4

2) In ardupilot_gazebo, navigate to the delta_wing.sdf file that will be in the models subfolder. You should see the ardupilot plugin in here under a control tag. Modify these contents to have the following: 

<jointName>zephyr_delta_wing::propeller_joint</jointName>
        <useForce>1</useForce>
        <multiplier>838</multiplier>
        <servo_min>1100</servo_min>
        <servo_max>1900</servo_max>
        <offset>0</offset>
        <type>VELOCITY</type>
        <p_gain>0.2</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.5</cmd_max>
        <cmd_min>0.0</cmd_min>
      </control>
      <control channel="1">
        <jointName>zephyr_delta_wing::flap_right_joint</jointName>
        <useForce>1</useForce>
        <multiplier>1.048</multiplier>
        <offset>-0.5</offset>
        <servo_min>1100</servo_min>
        <servo_max>1900</servo_max>
        <type>POSITION</type>
        <p_gain>10.0</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.5</cmd_max>
        <cmd_min>-2.5</cmd_min>     
      </control>
      <control channel="0">
        <jointName>zephyr_delta_wing::flap_left_joint</jointName>
        <multiplier>1.048</multiplier>
        <offset>-0.5</offset>
        <servo_min>1100</servo_min>
        <servo_max>1900</servo_max>
        <type>POSITION</type>
        <p_gain>10.0</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.5</cmd_max>
        <cmd_min>-2.5</cmd_min> 

After these changes, launch the delta wing world file with gazebo. Then launch a SITL instance. If you set up a waypoint mission and switch to auto mode, after arming the aircraft will take off and perform the mission. 

Can you post these changes to your main read me? 

This resource was good: https://discuss.ardupilot.org/t/ignition-gazebo-sitl-support/76120

SwiftGust, 2021-12-01:
@NDevDrone Thank you for detailed review and sorry for inconvenience.

For the delta_wing.sdf change doesn't apply at all since ArduPilotPlugin itself doesn't have servo min & max SDF parameters,
control packet coming from ArduPilot comes in as normalized from 0.0 to 1.0 range, not a PWM value.

To clarify this issue, core issue is left and right elevon channel set to ArduPlane and Gazebo Delta-wing model is flipped somehow.

You can just correct this either simple way rather than modify all as you suggested
**Method 1 - by changing ArduPlane Parameter**
```
SERVO1_FUNCTION = 77 -> 78 (Elevon Left -> Elevon Right)
SERVO2_FUNTION = 78 -> 77 (Elevon Right -> Elevon Left)
```
**Method 2 - by chaning zephyr model channel mapping**
```
      **<control channel="0">**
        <multiplier>-1</multiplier>
        <offset>-0.5</offset>
        <type>POSITION</type>
        <p_gain>10.0</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.0</cmd_max>
        <cmd_min>-2.0</cmd_min>
        <jointName>zephyr_delta_wing::flap_left_joint</jointName>
      </control>
      **<control channel="1">**
        <multiplier>-1</multiplier>
        <offset>-0.5</offset>
        <type>POSITION</type>
        <p_gain>10.0</p_gain>
        <i_gain>0</i_gain>
        <d_gain>0</d_gain>
        <i_max>0</i_max>
        <i_min>0</i_min>
        <cmd_max>2.0</cmd_max>
        <cmd_min>-2.0</cmd_min>
       <jointName>zephyr_delta_wing::flap_right_joint</jointName>
      </control>
```

Thank you for digging into this, and I wasn't aware of the this issue.
For the last time I checked, it flew fine without these changes

To this issue resolved, I think it is best to apply method 2 to this repository.
since default ArduPlane SITL parameter change for zephyr model is need to be incorporated in https://github.com/ArduPilot/ardupilot

NDevDrone, 2021-12-01:
@SwiftGust Please do not sweat it, no need to apologize for some inconvenience there was none! You've provided an excellent resource here and are a valuable member of the ardupilot community. Thank you very much for your time and effort, and still helping with issues after all this time. 

It was a joy to get the zephyr delta wing flying with SITL and having multiple flying around. There aren't really any other good solutions for simulating ArduPlane with Gazebo as far as I've found.

SwiftGust, 2021-12-04:
@NDevDrone Thank you for such kind words.

For the repository alone, There are lots to go on in compared to https://github.com/PX4/PX4-SITL_gazebo, 
need to incorporate more flight vehicle models and need support for Gazebo ROS and sensors so on.

closed 2021-12-04 by SwiftGust