SwiftGust/ardupilot_gazebo #44: parse as sdf version 1.7 failed with Gazbo11

mrherman, 2022-01-04:
When running `gazebo --verbose zephyr_ardupilot_demo.world` I get the following error:
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.109.41.239
Error [parser.cc:525] parse as sdf version 1.7 failed, should try to parse as old deprecated format
Error Code 4 Msg: Required attribute[xmlns:xacro] in element[sdf] is not specified in SDF.
Error Code 8 Msg: Unable to parse sdf element[sdf]
[Msg] Loading world file [/home/hermamr1/ardupilot_gazebo/worlds/zephyr_ardupilot_demo.world]
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.109.41.239
```

When running `gazebo --verbose /usr/share/gazebo-11/worlds/zephyr_demo.world` I do not get a similar error.

I start with a fresh Ubuntu 20.04 install, and added the ros package list and installed gazebo11 via
```
sudo apt install gazebo11
sudo apt install gazebo11-dev
```
I followed the install instructions on this git's readme installing/making from master.  During the cmake step I do get this warning (not sure if relevent):
```
-- Looking for ignition-common3 -- found version 3.14.0
-- Looking for ignition-math6 -- found version 6.9.2
-- Looking for ignition-msgs5 -- found version 5.8.1
Gazebo version: 11.9
-- Configuring done
WARNING: Target "ArduCopterPlugin" requests linking to directory "/home/hermamr1/ardupilot_gazebo/build".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "ArduPilotPlugin" requests linking to directory "/home/hermamr1/ardupilot_gazebo/build".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "ArduCopterIRLockPlugin" requests linking to directory "/home/hermamr1/ardupilot_gazebo/build".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/hermamr1/ardupilot_gazebo/build
```

Not sure if there is a missing step in the install instructions where I need to resolve these.

When doing the above but with Gazebo9 (don't need to add the ros package as gazebo9 in the default ubuntu package manager) I do not get the sdf parse error and things run as expected.

zeynelgokcek, 2023-01-02:
I get the same errors to did you fix it?

closed 2023-08-27 by mrherman

pratikPhadte, 2024-02-11:
I have the same error, were you able to resolve it?

zeynelgokcek, 2024-02-11:
No mate, I was using ubuntu 22 and it was not working. You can run it by
ubuntu 20 or 18.

şunu yazdı:

rookie330, 2025-03-10:
> I have the same error, were you able to resolve it?

rookie330, 2025-03-10:
I can import the model after deleting xmlns:xacro="http://www.ros.org/wiki/xacro" in the sdf tag