SwiftGust/ardupilot_gazebo #51: Multi-Vehicle Simulation (iris and zephyr)

amirrezasadeqi, 2022-07-19:
Hello Dear @SwiftGust 
I want to simulate an iris and a zephyr in the same world and control them with ardupilot flight-stack. So I copied the iris model into [zephyr demo world](https://github.com/SwiftGust/ardupilot_gazebo/blob/master/worlds/zephyr_ardupilot_demo.world). After that in two separated clone of ardupilot I ran bellow commands:

```
sim_vehicle.py -v ArduPlane -f gazebo-zephyr -m --mav10 --console -I0
sim_vehicle.py -v ArduCopter -f gazebo-iris -m --mav10 --console -I1
```
then I ran the gazebo with the changed world file, but the the iris quadrotor can't connect to it's SITL instance. In iris console, the message "link 1 down" is printed and in the terminal where gazebo is running, bellow line is printed:
```
[Err] [ArduPilotPlugin.cc:962] [iris] failed to bind with 127.0.0.1:9002 aborting plugin.
```
so could you please help me to find out where I is my fault and how I can connect each UAV to its SITL simulation?

SwiftGust, 2022-07-20:
Hi @amirrezasadeqi 
Have you changed port number in SDF file as well As it is stated in https://github.com/SwiftGust/ardupilot_gazebo#multi-vehicle-simulation?

amirrezasadeqi, 2022-07-21:
Hi there!
No. I did not do that, since I thought it would be done automatically and I just need to run the `sim_vehicle.py` with proper arguments.

I looked through out the SDF files to see how to change input/output ports and I find the `<fdm_port_in/out>` in the 'models/iris_with_standoffs_demo/model.sdf` file. So I changed them base on the read-me file and it works fine(Actually Iris does not arm sometimes but the connections are OK!).

But could you please help me to know how to override these parameters in another custom world file in which I've included the `iris_with_standoff_demo` like bellow:
```
<model name="iris">
      <pose> 0 0 0 0 0 0 </pose>
      <include>
        <uri>model://iris_with_standoffs_demo</uri>
         <pose> 0 0 0 0 0 0 </pose>
      </include>
    </model>
```
because I could not find anything online about how to do this and changing the original file is not very nice.

amirrezasadeqi, 2022-07-23:
Hi

Thanks for your answers!

Actually I don't need any auto-thing. I want to know if there is some way that I can use to override these options when I include the model in my world file, just like we give `<pose></pose>` in bellow chunk of code.
```
<model name="iris">
      <pose> 0 0 0 0 0 0 </pose>
      <include>
        <uri>model://iris_with_standoffs_demo</uri>
         <pose> 0 0 0 0 0 0 </pose>

          <--! Like bellow lines -->
         <fdm_port_in>9012</fdm_port_in>
         <fdm_port_out>9013</fdm_port_out>

      </include>
</model>
```
I used above code but SDF does not support it! Since I'm not very expert in SDF, I think I added it in wrong way. I searched the internet to find the right way of `overriding parameters when including a model file`, but I could not find anything helpful. So My `Question` is:

`Is there some kind of mechanism like above chunk of code for overriding the fdm_ports while including the iris model in my custom world file?`

If there is a way of overriding parameters, so we don't need to hard copy and paste a whole iris model file to just change two lines of it or we don't need to change them in the original file and in this way we just change theme when we include the iris model in our custom world file.

Forgive me for long stories!

Thanks!

SwiftGust, 2022-07-26:
Hi,
AFAIK, For SDF alone it doesn't support such, you should hard copy paste.
On the other hand, you can convert it using xacro->URDF->SDF is available method.

amirrezasadeqi, 2022-09-08:
Hi again
Dear @SwiftGust, Now I have another question and I hope you can help me.
I run SITL simulation for both iris and zephyr in the same location through `-L` option in `sim_vehicle.py` command. So I can assume when I give a `guided lat lon alt` command with the same arguments to both drones, the drones move toward a same location in the gazebo simulation, but each of them goes to a different place.

I tried to set the same `yaw` values for drones, but it does not help and still get the same behavior. Also I have read somethings about `WGS84` but I don't think it is important in my problem because it is the location of the world origin and the other things would have relative locations.

Also Note that I didn't write the world file from scratch and I have just created a copy of zephyr demo world with SDF version changed to  1.7, then opened it in the gazebo and finally added the iris model through gazebo gui and saved it. I don't know if this is important but I thought it may be important!

Sincerely!

amirrezasadeqi, 2022-10-24:
Hi Dear @SwiftGust

I think the problem caused by different coordinate frame conventions used for each drone. I just copied below lines in zephyr model and pasted it in the corresponding part of iris drone model and now both of drones go to the same spot in the gazebo world for the same target GPS location.
```    
 <modelXYZToAirplaneXForwardZDown>0 0 0 3.141593 0 -1.57079</modelXYZToAirplaneXForwardZDown>

 <gazeboXYZToNED>0 0 0 3.141593 0 -1.57079</gazeboXYZToNED>
```

I don't know if this was the correct solution or not, because the iris model some times does weird things like turning around itself(but becomes stable after a moment) but the overall behaviors are OK.

I think, now I should close this issue, but at the end could you guide me or give me some references to understand these coordinate frame conventions which are converting to each other.

Thanks for your helps ...